bigtreetech/BTT-Pi #7: BTT Pi v1.2 case .stl

claidheamohmor, 2023-05-28:
Sajnos a mellékelt kinyomtatható ház elég sok sebből vérzik, a HDMI, audio és tápcsatlakozók lyukainál a "bottom" és "middle" szekcióban kívűl mélyítést kellett készítenem, hogy a csatlakozók műanyag foglalatai elférjenek és a csatlakozók helyesen üljenek be az aljzatokba, illetve az alaplapi JI5 csatlakozó mellyett belül kivágást kellett csinánom, mert vastag a burkolat és nem fér el. A "bottom" fájban az USB csatlakozó melletti bepattintós lyuknál hibás a szerkesztés, illetve a "middle" szekciónak végül a teljes belsejét kivágtam, mert a hűtőborda miatt nem fér el. Sajnos a chip eléggé melegszik, így mindenképpen szükséges a hűtőborda, sőt úgy tűnik, a ventilátort sem lehet megspórolni. Ha valakit érdekel, a módosításokat feltöltöm a Thingiverse-re.

nickdollimount, 2023-06-23:
I also had to make similar modifications. I did the middle cutout for the heatsink as mentioned above, though I didn't have any of the other issues mentioned. The biggest issue for me is a section next to the SD card reader that sits right above one of the tiny components. It wasn't closing right for me so of course my first instinct was to squeeze harder; this piece broke the solder joint of the tiny component. I had to clip out that section of plastic and resolder the component. See the screenshot attached where I circled the offending piece.
[image]

[BTT Pi Case_Edit.zip](https://github.com/bigtreetech/BTT-Pi/files/11850265/BTT.Pi.Case_Edit.zip)

Escrich, 2024-06-29:
Try mine:

https://www.printables.com/model/927675-caja-para-btt-pi-btt-pi-box-frontal-front