bigtreetech/BTT-Pi #9: Connecting and spi configuration of SPI ADXL345 on the BTT-PI

Skappa75, 2023-06-10:
Hi, there is a lot of documentation on how to connect an ADXL345 on a CB1 or CM4, and their configuration on Klipper. But I'm in trouble when I connect the BTT-ADXL345 to the BTT-PI by the supplied cable on the dedicated SPI port (page 7 of its manual).
In the pin out what is the right cs_pin number?

Kimmy2000, 2023-06-29:
Hi, this is the setting you require for the BTT-ADXL345 to the BTT-PI 

[mcu host]
serial: /tmp/klipper_host_mcu

[adxl345] #BTT PI
cs_pin: host:gpio67
spi_bus: spidev0.0
axes_map: x,y,z

[resonance_tester]
accel_chip: adxl345
accel_per_hz: 70
probe_points: 117.5,117.5,10 

In the boardenv.txt uncomment only
overlays=spidev0_0

Skappa75, 2023-06-30:
Hi, thanks for your suggestions but unfortunately it doesn't work by
connecting it via the dedicated ADXL345 port as shown in the BTT-PI manual.
You wrote "cs_pin: host:gpio67" and I noticed that it is from the
40pin-GPIO PC3, you confirm that it is also the right cs.pin of the
dedicated ADXL345 port? Thanks for your help
Skappa

Skappa75, 2023-06-30:
I allways recive that error:
"adxl345: spi pins must be on same mcu"

Kimmy2000, 2023-06-30:
have you uncommented the SPI like i said here?

In the boardenv.txt uncomment only
overlays=spidev0_0

HYK-HideYourKey, 2023-08-02:
not working for me either... commented spidev0_0 already

UmitBaba, 2023-08-19:
[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:gpio67       # I m using fluidd, host:gpio67 doesn't allowed me to write so I tried rpi:gpio67 and just worked.
spi_bus: spidev0.0

[resonance_tester]
accel_chip: adxl345
probe_points:
    155, 155, 20

I m using btt pi v1.2 and this script worked for me. There is one important step, the sd card which plug in to btt pi, you need to plug it your pc and ın the boardenv.txt uncommend overlays=spidev0_0  (remove "#" sign) then dedicated adxl345 port on btt pi works.

Skappa75, 2023-08-19:
Very thanks, I'll try as soon is possible.

tiagomaricate, 2024-01-23:
What is the name/model of connector in the BTT Pi for the ADXL?

horapet, 2025-04-15:
> Hi, thanks for your suggestions but unfortunately it doesn't work by
> connecting it via the dedicated ADXL345 port as shown in the BTT-PI manual.
> You wrote "cs_pin: host:gpio67" and I noticed that it [...]