bigtreetech/BTT-Pi #11: Enabling the U2C can adapter

orotecluc, 2023-06-28:
I have bought a BTT-Pi and the U2C pinboard,bt when I assemble them, the PI does not find the canbus...
Is there somewhere an explanation on how to enable the U2C pinboard?

Byourei, 2023-07-06:
Hi, I have a similar problem. I bought a U2C module for the BTT Klipper Pi. Pi does not see U2C when connected, but the LEDs light up. When turned on with BOOT pressed, it can be seen in usb devices, can be flashed. and that's all. we did not find a guide or a manual anywhere. and technical support doesn't know either.

pauluchin, 2023-08-17:
Hi, I also have questions about how to use it. If you could provide more information on how to enable and use it, there isn't much information on the web.

DrKlipper, 2023-09-12:
I had the same issue. Just compile and flash a new firmware and it should work: 

- `sudo apt install cmake gcc-arm-none-eabi -y`
- `cd ~`
- `git clone --depth=1 -b stm32g0_support https://github.com/bigtreetech/candleLight_fw`
- `cd ~/candleLight_fw`
- `mkdir build`
- `cd build`
- `cmake .. -DCMAKE_TOOLCHAIN_FILE=../cmake/gcc-arm-none-eabi-8-2019-q3-update.cmake`
- `make budgetcan_fw`
bring CAN Adapter in DFU Mode ... Attach to USB C Cable, hold BOOT button and connect ... 
- Flash Adapter : `dfu-util -R -a 0 -s 0x8000000:mass-erase:force -D ~/candleLight_fw/build/budgetcan_fw.bin`

Hoeby, 2024-06-30:
Just to be sure.

Is this for the can adapter v1.0 which is mounted on top of the btt pi? Or for the bigger u2c can adapter V2.1?

orotecluc, 2024-06-30:
For the small v1 that goes on top of pi v1

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

Hoeby, 2024-06-30:
i ask it here, because there is not much information to find online about this u2c boad.

When i run "lsusb", then i find "OpenMoko, Inc. Geschwister Schneider CAN adapter". It finds the board.
When i run "~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0", then i don't see an ID from this board, is this correct? I would say every canbus device must have an ID, also this can adapter.

The reason why, i tried multi can boards to run on the can adapter, but nothing works. Looks like the issue is in the can adapter, but i am not sure

Mati989, 2025-01-17:
Hi. Can someone write what is the full process of flashing this small adapter that goes to the top of pi v1.2 please.I can't find any manual for this adapter.

hiaselhans, 2025-01-23:
@Mati989 
https://drklipper.de/doku.php?id=sbcs:61_-_bigtreetech_pi_v1.2_mit_can