bigtreetech/BTT-Pi #13: i2c overlay

zuverck, 2023-07-31:
Trying to determine which pins on the GPIO are tied to the i2c interface.  There is an overlay and i2c device defined in the BoardEnv.tst file but don't see any documentation or notation in the pins diagram on which pins these are connected to on the GPIO header.

intplus10, 2023-10-17:
It is matching with the standard Pi. I would refer to BTT-Pi user manual page 7, I hope you are searching for this information:
[image]

EidenPride, 2024-09-08:
> Trying to determine which pins on the GPIO are tied to the i2c interface. There is an overlay and i2c device defined in the BoardEnv.tst file but don't see any documentation or notation in the pins diagram on which pins these are connected to on the GPIO header.

Hi did you find out something about l2c?

BTTJUSTIN, 2024-09-09:
Thanks for making this suggestion. We will discuss it internally and update the documentation if it is approved

closed 2024-10-18 by BTTJUSTIN