bigtreetech/BTT-Pi #15: Btt pi not working on second day

cheuhyakuroku, 2023-08-31:
Mthe first day it run normal. I config klipper, and print some model file. But second day it not work. Power led is light but nothing happen. Device not turn on,

MarioPerini, 2023-12-22:
same here, waiting for response through mail support

Khalid310, 2024-01-19:
Have you both achieved any solution?

Mine was powered by 24v. The first time its green light was on for a while and then it was suddenly off. Then, each time I power it up, the green light blinks once and totally stops. Tested powering via USB, and it was OK.

MarioPerini, 2024-01-25:
No they declined my refund request on AliExpress. I will. Not buy any BTT products anymore, since I was treated like this. Mine run only for one day at 24V

Escrich, 2024-06-29:
Mine is working since a couple of months ago, without any problem, but I think it's a 5 V fan, not to sure about it now