bigtreetech/BTT-Pi #16: Fan pin issue

cheuhyakuroku, 2023-09-09:
anyone help me pls 🥹
btt pi v1.2 fan pin issue
[image]
[image]

Escrich, 2024-06-29:
[mcu BTT-Pi]
serial: /tmp/klipper_host_mcu

[temperature_fan BTT-Pi_CPU]  # Temperatura interna del procesador de la BTT Pi
pin: BTT-Pi:gpio211
max_power: 1.0
shutdown_speed: 0
sensor_type: temperature_host
target_temp: 40
max_temp: 85
min_temp: 5
control: pid
pid_kp: 1.0
pid_ki: 0.5
pid_kd: 2.0
off_below: 0.1 # Mantener apagado si la potencia aplicada está por debajo de este valor

Escrich, 2024-06-29:
Check the names you used