bigtreetech/BTT-Pi #20: Enable 1-wire

Puzzl88, 2023-09-27:
How do you enable 1-wire on the BTT-Pi, specifcally for the DS18B20 sensor?

mdvorak, 2023-09-29:
I've created guide how to do that:
https://gist.github.com/mdvorak/2fccca3fa9f76f5a2e11e567797c76d6

Please let me know if it works, or if the guide needs some improvements/fixes. Thanks

Duplicate to https://github.com/bigtreetech/CB1/issues/65 and https://github.com/bigtreetech/CB1/issues/104

Puzzl88, 2023-09-30:
> Please see [bigtreetech/CB1#65 (comment)](https://github.com/bigtreetech/CB1/issues/65#issuecomment-1740552051) for the answer.

biqu@BTT-CB1:~/w1-gpio-cl$ sudo make install
install -m644 w1-gpio-cl.ko /lib/modules/`uname -r`/kernel/drivers/w1/masters
install: cannot stat 'w1-gpio-cl.ko': No such file or directory
make: *** [Makefile:95: install] Error 1

mdvorak, 2023-09-30:
Hmm, maybe I forgot a package, I'll take a look, thanks

Puzzl88, 2023-10-01:
> Hmm, maybe I forgot a package, I'll take a look, thanks

I understood what's the problem.
After LN -s/Lib/Modules/`Uname -r`/Build/Include/Linux W1, you need to execute a command make.

Thanks for the help!

mdvorak, 2023-10-01:
ah, great, I'll update the guide, thanks!

wmetnitz, 2023-10-03:
sorry, i'm a unix newbe - i get the following error:
---
biqu@voron:~/w1-gpio-cl$ **sudo make**
NOTE: ./w1 symlink is already set and will not be updated. Remove it and restart the compilation process in case you want to re-run the kernel sources examination.
NOTE: gen-mast.h was generated
make -C /lib/modules/5.16.17-sun50iw9/build M= modules
make[1]: Entering directory '/usr/src/linux-headers-5.16.17-sun50iw9'
make[2]: *** No rule to make target 'arch/arm64/tools/gen-cpucaps.awk', needed by 'arch/arm64/include/generated/asm/cpucaps.h'.  Stop.
make[1]: *** [arch/arm64/Makefile:173: archprepare] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.16.17-sun50iw9'
make: *** [Makefile:13: all] Error 2

mdvorak, 2023-10-03:
Hmm, that is wierd error :-)

mdvorak, 2023-10-04:
I've managed to reproduce it with latest packages, maybe it was caused by some recent system update, I'll investigate it.

mdvorak, 2023-10-04:
@wmetnitz Found it.
`make` needs to be run without `sudo`, for some reason, it messes up env vars and compilation fails to find some files. Also for the compilation itself, sudo should not be really needed.
Guide updated.

_note: `sudo -E make` seems to work file too.`

mdvorak, 2023-10-04:
That was entirely mistake on my side, sorry!
```
echo 'options w1-gpio-cl m1="gdt:71"' | sudo tee /etc/modprobe.d/w1-gpio-cl.conf
echo 'w1-gpio-cl' | sudo tee /etc/modules-load.d/w1-gpio-cl.conf
```
I needed a tester, thanks

wmetnitz, 2023-10-04:
Hi Michal,
thanks, but it doesn't work:

biqu@voron:~/w1-gpio-cl$ echo 'options w1-gpio-cl m1="gdt:71"' | sudo tee /etc/modprobe.d/w1-gpio-cl.conf
options w1-gpio-cl m1="gdt:71"
biqu@voron:~/w1-gpio-cl$ echo 'w1-gpio-cl' | sudo tee /etc/modules-load.d/w1-gpio-cl.conf
w1-gpio-cl
biqu@voron:~/w1-gpio-cl$ sudo modprobe w1-gpio-cl
modprobe: ERROR: could not insert 'w1_gpio_cl': Exec format error
biqu@voron:~/w1-gpio-cl$

wmetnitz, 2023-10-04:
Hi, thanks for your help, but it doesn’t work…

 

 

***@***.***:~/w1-gpio-cl$ echo 'options w1-gpio-cl m1="gdt:71"' | sudo tee /etc/modprobe.d/w1-gpio-cl.conf

options w1-gpio-cl m1="gdt:71"

***@***.***:~/w1-gpio-cl$ echo 'w1-gpio-cl' | sudo tee /etc/modules-load.d/w1-gpio-cl.conf

w1-gpio-cl

***@***.***:~/w1-gpio-cl$ sudo modprobe w1-gpio-cl

modprobe: ERROR: could not insert 'w1_gpio_cl': Exec format error

***@***.***:~/w1-gpio-cl$

math-milan, 2025-01-02:
Hi, I have trouble finding the kernel headers. I have used the new release with the 6.6.66 Kernel but i only find legacy headers with apt. Do i have to get the headers from the kernel direktly ( from source )?

I am trying this at the moment but so far no luck. Thangs for sugestions