bigtreetech/BTT-Pi #22: ethernet bouncing

tonhe, 2023-10-25:
```
[  746.806264] sunxi-gmac 5030000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[  794.934660] sunxi-gmac 5030000.ethernet eth0: Link is Down
[  796.982767] sunxi-gmac 5030000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[  826.679038] sunxi-gmac 5030000.ethernet eth0: Link is Down
[  827.703084] sunxi-gmac 5030000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[  833.847133] sunxi-gmac 5030000.ethernet eth0: Link is Down
[  835.895247] sunxi-gmac 5030000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off

```
This is constant. I saw other folks having issues similar on the armbian forums.

christian-nils, 2023-11-14:
Same issue here.

dworki, 2023-11-15:
And here too.

iHat3Microsoft, 2026-08-19:
same issue. tried mainline armbian and it has same issue, faulty hardware?