bigtreetech/BTT-Pi #23: launch BC1 on BTT PI from USB

stepand70, 2023-10-29:
Is it possible to teach BTT Pi to run from USB?

idontknowtom, 2023-11-30:
@stepand70 You mean to run from a USB drive or SSD?

stepand70, 2023-11-30:
> @stepand70 You mean to run from a USB drive or SSD?

USB

idontknowtom, 2023-11-30:
@stepand70 You have to setup with a micro SD card first. Then transfer system over to USB. But the BOOT partition will stay on the micro SD card so you need to keep the card in the BTT Pi. The CB1 and BTT Pi are essentially clones of the Orange Pi Zero 2.

So follow this [guide](https://jamesachambers.com/orange-pi-zero-2-usb-ssd-boot-guide/) and it will work. You won’t find any better performance with a USB thumb drive. In fact the guide says you’ll get worse performance. I tested as well. USB to M.2 SATA SSD works amazing.

stepand70, 2023-11-30:
> @stepand70 You have to setup with a micro SD card first. Then transfer system over to USB. But the BOOT partition will stay on the micro SD card so you need to keep the card in the BTT Pi. The CB1 [...]

thanks for the advice! It's not about performance, it's not convenient to take out the SD card when it's in the case, I make copies of CB1 and klipper every 3 days))