bigtreetech/BTT-Pi #24: Display not working

killrob, 2023-12-09:
Hi all,
i have a PTTPi and a TFT35SPI V2.1, i've connected them with the ribbon cable provided with  monitor.
Klipper and Klipperscreen are installed but the display is all white.
what i'm doing wrong?
this is an extract from log:
`2023-12-09 14:17:20,514 [functions.py:log_start()] - --------------------KlipperScreen Log Start--------------------
2023-12-09 14:17:20,516 [functions.py:log_start()] - KlipperScreen Version: v0.3.7-10-gdd3b255
2023-12-09 14:17:20,516 [functions.py:log_start()] - Python version: 3.9
2023-12-09 14:17:20,935 [config.py:get_config_file_location()] - Passed config (-c): /home/biqu/KlipperScreen.conf
2023-12-09 14:17:20,937 [config.py:__init__()] - Config path location: /home/biqu/KlipperScreen/ks_includes/defaults.conf
2023-12-09 14:17:21,004 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2023-12-09 14:17:21,348 [config.py:create_translations()] - Selected lang: None OS lang: en_US
2023-12-09 14:17:21,350 [config.py:install_language()] - Using system lang
2023-12-09 14:17:21,352 [config.py:install_language()] - Using lang en
2023-12-09 14:17:21,436 [screen.py:__init__()] - Screen resolution: 1024x768
2023-12-09 14:17:21,442 [KlippyGtk.py:__init__()] - Font size: 23.8 (medium)
2023-12-09 14:17:24,534 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2023-12-09 14:17:24,548 [screen.py:set_screenblanking_timeout()] - Using DPMS
2023-12-09 14:17:24,558 [screen.py:initial_connection()] - Default printer: None
2023-12-09 14:17:24,559 [screen.py:connect_printer()] - Connecting to printer: Printer
2023-12-09 14:17:24,559 [screen.py:_load_panel()] - Loading panel: splash_screen
2023-12-09 14:17:24,705 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2023-12-09 14:17:24,721 [KlippyWebsocket.py:connect()] - Attempting to connect
2023-12-09 14:17:24,760 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-12-09 14:17:24,782 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2023-12-09 14:17:24,784 [KlippyWebsocket.py:connect()] - Starting websocket thread
2023-12-09 14:17:24,800 [_logging.py:info()] - Websocket connected
2023-12-09 14:17:24,801 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2023-12-09 14:17:25,096 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-12-09 14:17:25,117 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-12-09 14:17:25,119 [screen.py:init_printer()] - Moonraker info {'klippy_connected': False, 'klippy_state': 'disconnected', 'components': ['klippy_connection', 'application', 'websockets', 'internal_transport', 'dbus_manager', 'database', 'file_manager', 'klippy_apis', 'secrets', 'template', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'http_client', 'announcements', 'webcam', 'extensions', 'octoprint_compat', 'authorization', 'history', 'update_manager'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes'], 'warnings': [], 'websocket_count': 1, 'moonraker_version': 'v0.8.0-206-g9ad8fed', 'missing_klippy_requirements': [], 'api_version': [1, 4, 0], 'api_version_string': '1.4.0'}
2023-12-09 14:17:25,138 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/webcams/list
2023-12-09 14:17:25,141 [printer.py:configure_cameras()] - Cameras: []
2023-12-09 14:17:25,142 [screen.py:init_printer()] - Klipper not connected
2023-12-09 14:17:28,149 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-12-09 14:17:28,168 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-12-09 14:17:28,170 [screen.py:init_printer()] - Moonraker info {'klippy_connected': False, 'klippy_state': 'disconnected', 'components': ['klippy_connection', 'application', 'websockets', 'internal_transport', 'dbus_manager', 'database', 'file_manager', 'klippy_apis', 'secrets', 'template', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'http_client', 'announcements', 'webcam', 'extensions', 'octoprint_compat', 'authorization', 'history', 'update_manager'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes'], 'warnings': [], 'websocket_count': 1, 'moonraker_version': 'v0.8.0-206-g9ad8fed', 'missing_klippy_requirements': [], 'api_version': [1, 4, 0], 'api_version_string': '1.4.0'}
2023-12-09 14:17:28,187 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/webcams/list
2023-12-09 14:17:28,188 [printer.py:configure_cameras()] - Cameras: []
2023-12-09 14:17:28,189 [screen.py:init_printer()] - Klipper not connected
2023-12-09 14:17:31,150 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-12-09 14:17:31,169 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-12-09 14:17:31,171 [screen.py:init_printer()] - Moonraker info {'klippy_connected': False, 'klippy_state': 'disconnected', 'components': ['klippy_connection', 'application', 'websockets', 'internal_transport', 'dbus_manager', 'database', 'file_manager', 'klippy_apis', 'secrets', 'template', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'http_client', 'announcements', 'webcam', 'extensions', 'octoprint_compat', 'authorization', 'history', 'update_manager'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes'], 'warnings': [], 'websocket_count': 1, 'moonraker_version': 'v0.8.0-206-g9ad8fed', 'missing_klippy_requirements': [], 'api_version': [1, 4, 0], 'api_version_string': '1.4.0'}
2023-12-09 14:17:31,189 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/webcams/list
2023-12-09 14:17:31,191 [printer.py:configure_cameras()] - Cameras: []
2023-12-09 14:17:31,191 [screen.py:init_printer()] - Klipper not connected
2023-12-09 14:17:34,155 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-12-09 14:17:34,178 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-12-09 14:17:34,181 [screen.py:init_printer()] - Moonraker info {'klippy_connected': True, 'klippy_state': 'startup', 'components': ['klippy_connection', 'application', 'websockets', 'internal_transport', 'dbus_manager', 'database', 'file_manager', 'klippy_apis', 'secrets', 'template', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'http_client', 'announcements', 'webcam', 'extensions', 'octoprint_compat', 'authorization', 'history', 'update_manager'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes'], 'warnings': [], 'websocket_count': 1, 'moonraker_version': 'v0.8.0-206-g9ad8fed', 'missing_klippy_requirements': [], 'api_version': [1, 4, 0], 'api_version_string': '1.4.0'}
2023-12-09 14:17:34,202 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/webcams/list
2023-12-09 14:17:34,204 [printer.py:configure_cameras()] - Cameras: []
2023-12-09 14:17:34,241 [KlippyRest.py:_do_request()] - 404 Client Error>>Not Found for url>>http://127.0.0.1:7125/printer/info
2023-12-09 14:17:37,147 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-12-09 14:17:37,168 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-12-09 14:17:37,170 [screen.py:init_printer()] - Moonraker info {'klippy_connected': True, 'klippy_state': 'startup', 'components': ['klippy_connection', 'application', 'websockets', 'internal_transport', 'dbus_manager', 'database', 'file_manager', 'klippy_apis', 'secrets', 'template', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'http_client', 'announcements', 'webcam', 'extensions', 'octoprint_compat', 'authorization', 'history', 'update_manager'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes', 'config_examples', 'docs'], 'warnings': [], 'websocket_count': 1, 'moonraker_version': 'v0.8.0-206-g9ad8fed', 'missing_klippy_requirements': [], 'api_version': [1, 4, 0], 'api_version_string': '1.4.0'}
[... cut ...]
2023-12-09 14:17:37,925 [main_menu.py:update_graph_visibility()] - Could not create graph tempstore: []
2023-12-09 14:17:43,147 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/temperature_store
2023-12-09 14:17:43,172 [printer.py:init_temp_store()] - Temp store: ['heater_bed', 'temperature_fan raspberry_pi', 'extruder']
2023-12-09 14:17:43,193 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/config
2023-12-09 14:17:43,195 [screen.py:init_tempstore()] - Temperature store size: 1200`

idontknowtom, 2023-12-10:
@killrob Are you using the full CB1 image or minimal CB1 image? Have you uncommented the `overlays=tft35_spi` in BoardEnv.text on the BOOT partition?

killrob, 2023-12-10:
hi @idontknowtom ,
i've downloaded the OS from here 
https://github.com/bigtreetech/CB1/releases/download/V2.3.2/CB1_Debian11_Klipper_kernel5.16_20230303.img.xz
and i don't know if is the minimal or the full one
this is my BoardEnv 
[BoardEnv.txt](https://github.com/bigtreetech/BTT-Pi/files/13627175/BoardEnv.txt)

idontknowtom, 2023-12-10:
@killrob I just downloaded and flashed that CB1 image and removed the # for tft35-spi overlay. Put the SD card in my BTT Pi with the screen attached and it booted into Klipperscreen first go.

Perhaps double check your wiring with the white flat FPC cable. Make sure it is fully seated in both connectors. The blue part points toward the BTT Pi heatsink and away from the PCB on the screen.

killrob, 2023-12-10:
@idontknowtom i've done exactly what you do, got a triple check on ribbon cable connection and is right connected as you told.
i'm waitig for another TFT35 SPI from aliexpress in few days, let's see if is a monitor or ribbon cable issue

killrob, 2023-12-10:
how much pin is that flat FCP cable?

killrob, 2023-12-10:
for everyone intrested,
i've solved in this way:
in BoardEnv
overlays=tft35_spi25
set speed at 25000 instead of 50000
commented
#overlays=spidev1_2

added
Identifier ""

to dpms.conf in /etc/X11/xorg.conf.d/

saved all, rebooted and now works