bigtreetech/BTT-Pi #25: Unable to run ADXL345 from 40 GPIO pins

Vicepredator, 2023-12-18:
Hi, i have a BTT-PI V1.2 and i'm trying to wire the ADXL345 sensor to the GPIO pins.

This is my configuration
```
[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: None
spi_bus: spidev1.2

[resonance_tester]
accel_chip: adxl345
probe_points: 117.5,117.5,10
```

I've wired the sensor as follows
```
VCC -> 3.3V (Yellow pin)
GND -> GND (Black pin)
CS -> PG12
SDO -> PH8
SDA -> PH7
SCL -> PH6
```

I've tried with all the spi_bus in BoardEnv.txt, but nothing, it doesn't work.

Any clue?

Tankpadua, 2024-01-13:
Why use the GPIO pins when the BTT Pi have the SPI port for the ADXL345 sensor?

Vicepredator, 2024-01-14:
Because I would like to use a standard Amazon sensor, and I don't wanna buy the specific connector.

Gurkenfisch, 2024-01-20:
Just connect it via the dedicated SPI port on the BTT-Pi I have just done that and it works.

mrvander18, 2024-02-20:
Yes, but how. There is no pinout that I can find that matches the BTT-Pi User Manual to the ADXL345.

User Manual ADXL345 Port Pinout
5V
GND
PC3
PC0
PC2
PC4

Board Pinout
GND
VCC
CS
INIT1
INIT2
SDO
SDA
SCL

Now the GND and VCC(5v) are easy enough and from GPIO pinout diagrams, one can deduce the INIT pins are ignored. But HOW do the PC pins map to CEO, Serial Data In, Out and Clock?

**User Manual >> Board**
5V >> VCC
GND >> GND
PC3 >> ??
PC0 >> ??
PC2 >> ??
PC4 >> ??

Amazon Generic Board:
https://www.amazon.com/gp/product/B0BXWHTXWT/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Gurkenfisch, 2024-02-20:
Ok so I have hooked it up via following:

5V >> VCC
GND >> GND
PC3 >> CS
PC0 >> SCL
PC2 >> SDA
PC4 >> SDO

And then in Klipper I have set it up like the following:

[adxl345] #BTT PI
cs_pin: host:gpio67
spi_bus: spidev0.0
axes_map: x,y,z

[resonance_tester]
accel_chip: adxl345
accel_per_hz: 70
probe_points: 60,60,10

I am almost certain that I just followed the normal input shaper install instructions from there. I am not 100% sure though. It works for me with that wiring.

Vicepredator, 2024-03-08:
Thanks, that seems good.

closed 2024-03-08 by Vicepredator

Hajto, 2024-05-28:
I am having the same issue. Connected the board as in the chart above. No luck klipper detecting the adxl tho