bigtreetech/BTT-Pi #33: Independent fan control

Escrich, 2024-06-29:
Dear all, I have one of my BTT Pi inside a Klipper machine, working with the code you provided to make the fan run, without any problem, but, I have a concern, when you are developing something, or simply you will not use this BTT-Pi for klipper, the fan remains stopped, and maybe the CPU will become hot, then, from my point of view, could be a great idea to have something similar as the Raspberry pi has, in order to control the fan in a regulated way, directly as a part of the OS, enclosed I send you three pictures in order to explain what I mean:

[image]

[image]

[image]

Escrich, 2024-07-03:
Just doing this control, will be a good idea to make a proportional control, instead of the simple on/off control that Raspberry have

PhilippeLafargue, 2024-09-17:
You are right. Actual solution is klipper dependent: fan can be controlled by klipper only. 

But fan control should be OS dependent, as it is related to the MCU cooling, klipper running or not. There are existing solutions for Raspberry PI that could be adapted to BTT-PI.